samphonic/MoppyClassic #161: Micro Period Calculations

dmadison, 2017-04-28:
I'm not sure if the repository is still being maintained, but I'm working on some expansions of the Moppy code and I'm wondering: how were the "microPeriods" for the notes determined? Is there a formula you used?

Thanks for the help!

SammyIAm, 2017-04-28:
Not a lot of maintenance / contributions from myself for quite a while, but there's always a bit of activity.

The microPeriods are the period (time between wave peaks) of the MIDI note in microseconds.  I used [this table](http://www.phy.mtu.edu/~suits/notefreqs.html) to determine the frequency for a particular note and followed this formula:

`(1 / <frequency>) * 1000000` or just `1000000 / <frequency>`

At MoppyPlayerOutput.java:84 , we're essentially dividing this period in half (so that the ticks happen twice as fast) because in the Arduino code each tick only toggles the state of the pin.  In other words, we send an `ON` and an `OFF` once per calculated period.

dmadison, 2017-05-01:
Awesome! Thanks for the quick response, that's all very clear.

closed 2017-05-01 by dmadison