samphonic/MoppyClassic #164: Software not working for 16 Drives?

AceoJack, 2017-05-26:
Hello There!
My name is Jack and I am year 9 student in Australia. I am doing a whole year project and I am currently stuck on a software problem. I believe the software is not working for 16 drives because I have 16 drives attached to 2 arduino UNO's and only 8 will ever work at one time. They will work across ardunios and I have modified the Midi file to have all 16 channels taken up by something. I am very in-experienced with arduino for this is my first ever project doing so. If someone is very kind and provide a 16 floppy, moppy software or tell me if I have something incorrect with my setup, that would be super cool and fantastic. I understand that I am a massive rooky and look like a complete loser because I lack the capabilities to even understand the code. I understand a little bit but I still have some problems. You guys do not need to feel obliged to help me but I am really stuck.  This is probably a easy task to do but I wouldn't even have a clue. You guys are awesome :D 

Many thanks and appreciation!
-Jack (AceoJack)

MoppyDoppy, 2017-05-26:
Hi
You are a bit right. If you control 16 channels with one note each there is a trick. The drivepool with roundrobin and stack uses all 16. 
my trick
You must change the code in the java application. The Arduino understand channel 0-7. so the Midi stream must change all channel 8-15 with -8. somewhere in the receiver class directly before sending out via serial.
currently I am on travel so I can't give you the code. You can see it working on my YouTube channel MoppyDoppy.

AceoJack, 2017-05-26:
Thanks for the help. I checked your Channel and I couldn't quite get anything from it. I'll search again. I don't really understand the terminology used here because I am very in-experienced. I understand if you are on a trip so don't worry about replying. If someone could send an updated Moppy.ino file, that would be superb! :D

SammyIAm, 2017-05-27:
Hi there!  Always happy to help, and there's no time like the present to dive in and start trying to understand the code.  That edited line you shared appears to be the correct solution (for only the second Arduino, but not the first one).  The `[index]` of the `currentPeriod` array is the Arduino pin associated to the drive you want to have play the note.  I.e. the first drive on the first Arduino would be `[2]`.  When you get up to the 9th drive, the index being sent will be `[18]`, but since you actually want to play the note on the *first* drive of the *second* Arduino, you'll want that Arduino to treat `[18]` as if it were `[2]` (hence the subtracting 16).  If you want to post your edited code to pastebin, I'd be happy to take a look at it.

If you've got both Arduinos connected to your PC, each will probably have its own COM port, so make sure in your MoppyUI that you've selected a different COM port for channels 9-16.  (Post a screenshot of your settings here and we can confirm they look correct)

Another possibility is that coincidentally the notes you've put into channels 9-16 are outside of the floppy's range.  Just to check, you may want to take one of the sample songs (e.g. for 3 drives) that you know works, and just copy those 3 channels over and over to make 16 channels of the same thing.  That way you can rule out of the MIDI files as the source of the problem.

AceoJack, 2017-05-28:
Hey there Sammy!
Here is the file I have probably edited horribly haha and also the Midi Test Clip I downloaded from SolidSnake (sorry if I butchered his name). If it is edited incorrectly, could you possibly send me the correctly edited file so I can test it out? The COM ports in the MoppyUI have been correct and I have changed them around multiple times, seeing if that is the problem. Another piece of info that could be useful is that I am using a USB cable extender for both arduinos. I can play the music fine and both COM 7 and 8 come up so I don't believe that is the problem. Thank you for the help and spending your time replying to me :D 

-Jack
(p.s the file has been zipped for both the midi and moppy.ino)
[16.Drive.Test.Clips.zip](https://github.com/SammyIAm/Moppy/files/1033912/16.Drive.Test.Clips.zip)
[Moppy_Jack_s_version.zip](https://github.com/SammyIAm/Moppy/files/1033913/Moppy_Jack_s_version.zip)

MoppyDoppy, 2017-05-29:
Hi
I used the same code on both arduino's, no change! 
I only changed the java code. That is much easier.

In class MoppyCOMBridge replace sendEvent with
--
public void sendEvent(byte pin, int periodData) {
        if (pin < 17)
        {
            sendEvent(pin, (byte) ((periodData >> 8) & 0xFF), (byte) (periodData & 0xFF));
        }
        else
        {
            sendEvent((byte)(pin-16), (byte) ((periodData >> 8) & 0xFF), (byte) (periodData & 0xFF));
        }
    }
--
Start the MoppyDesk and configure channel 1-8 to the first Moppy Serial COM Port and 9-16 to the second Moppy Serial COM Port.
--
Now you can use it with 16 drives

SammyIAm, 2017-05-29:
Hooray!  Glad you got it working.

closed 2017-05-29 by SammyIAm

AceoJack, 2017-05-29:
I'll test it as soon as I get home! I;m assuming the java code is in the moppy advanced folder? You guys are awesome! :DD

AceoJack, 2017-05-30:
Hello again! I cannot seem to locate the java code? Is it the moppy.ino or is it another file within the moppy-advanced folder? Can someone tell me the directory or the place where I can locate the java code. Thank you again and sorry if this is a super stupid question :D

AceoJack, 2017-05-30:
I believe I have found it but when I modify the code, it comes up with lots of errors asking me to edit and add params? I am using notepad++ to edit but I am still questioning if I am doing it right because you say it as its one or 2 copy and pastes. I am extremely confused? Thank you!

SammyIAm, 2017-05-31:
Hey, @AceoJack , I'm sorry I closed this too soon.  I somehow thought @MoppyDoppy 's reply was from you and that you'd got things working.  My apologies.  Sounds like you managed to get it working over in the other issue though!