samphonic/MoppyClassic #165: Java code editing?

AceoJack, 2017-05-30:
Hi all again!
Sorry to be spamming a little but I really do need some help here :/ 
I posted an issue 4 days ago needing help with the getting 16 drives to work. I got some helpful replies but I cant seem to edit the code successfully. Here is what MoppyDoppy posted and can someone or you, MoppyDoppy, please help with the edit of the code. I seem to be completely useless at this stuff. Sorry :( 

Hi
 I used the same code on both arduino's, no change!
 I only changed the java code. That is much easier.

In class MoppyCOMBridge replace sendEvent with
 public void sendEvent(byte pin, int periodData) { if (pin < 17) { sendEvent(pin, (byte) ((periodData >> 8) & 0xFF), (byte) (periodData & 0xFF)); } else { sendEvent((byte)(pin-16), (byte) ((periodData >> 8) & 0xFF), (byte) (periodData & 0xFF)); } } 
 Start the MoppyDesk and configure channel 1-8 to the first Moppy Serial COM Port and 9-16 to the second Moppy Serial COM Port.

Now you can use it with 16 drives

I think there are multiple MoppyCOMBridge files and I don't know which one I need to edit. Also, I think I am replacing the code into the wrong 'sendEvents' because I don't know how to do edit code properly. If this works for someone else, can you please zip the file and send it to me? That would be SUPER appreciated and helpful for I am stuck. You guys are actually awesome because you are so nice to someone so in-advanced in this field. Thank you!

MoppyDoppy, 2017-05-30:
Hi AceoJack,
(sorry of my english)
I have attached my complete moppy folder including java and arduino ino file. 
You should use NetBeans as Editor/IDE for java. There is only one MoppyCOMBrifge.java unter outputs. I have attached a screen shoot of my NetBeans IDE.

Just download it, exctarct and open the project MoppyDesk in Netbeans. It should compile. Maybe you must choose the available java sdk on your machine to compile it. 

[Moppy.zip](https://github.com/SammyIAm/Moppy/files/1037999/Moppy.zip)
[image]
[image]

AceoJack, 2017-05-30:
HAHA YES!! Thank you so much MoppyDoppy for your terrific help!! Everything works fine now which is fantastic! You are truly amazing for you help. You respond very quickly and help me even thought English isn't your first language. You are absolutely superb! Hope you have a good day/night! THANK YOU!

-Jack :DDDD

SammyIAm, 2017-05-31:
Thanks, @MoppyDoppy !  And sorry again, @AceoJack for closing and abandoning that other issue.  Glad it's working!  Come back and share a video maybe when you have it all up and running :)

closed 2017-05-31 by SammyIAm

AceoJack, 2017-06-01:
Will send a pic once it is done. Thanks again to everyone! You guys is what makes the internet a bearable place! Thank you :DDDDD