samphonic/MoppyClassic #166: Change MAX_POSITION[] variable

GeorgeTsakiridis, 2017-06-06:
Hello, I am sorry I created an issue, but I didn't know how to contact the author. Well I am into the "floppy drive music" hobby for some time and I tried experimenting with the MAX_POSITION[] variable changing it to only 2. When I tried to play some music I noticed that the notes were way more louder and the high frequency notes actually sounded more loud that the low ones. (under normal circumstances it goes the other way arround).  Of course the heads of the floppy drives weren't moving, just vibrating. I just wanted to ask, can this damage or reduce the life of the floppy drives? Has anyone tried it besides me? Any comment?
(From the Arduino code)
byte MAX_POSITION[] = {
  //0,0,158,0,158,0,158,0,158,0,158,0,158,0,158,0,158,0
 0,0,2,0,2,0,2,0,2,0,2,0,2,0,2,0,2,0
  };

SammyIAm, 2017-06-07:
While I'm not sure anyone knows *for sure*, I doubt it will significantly reduce the life of the floppy drives.  (Potentially it seems like it might put a *little* more stress on the motors because they may be having to fight a bit of momentum each time they change direction, but they do that anyway each time the head changes direction under normal operation so it's not like it's outside of their design specs)

Additional unsolicited thoughts: If you're feeling adventurous, and the notes are consistently louder, you could try switching between 2 and 158 based on the MIDI note's velocity (loudness).  It's not a *lot* of dynamic control, but could potentially make it easier to keep the melody louder than the background notes.

GeorgeTsakiridis, 2017-06-07:
Thank you very much for your reply. That's an interesting idea, setting to 2, midi notes that have high frequency and reverting back to 158 notes of low frequency.

closed 2017-06-07 by GeorgeTsakiridis