samphonic/MoppyClassic #168: License?

dmadison, 2017-06-23:
I know this has been brought up before (#53), but have you given any more thought to adding a formal license to the Moppy code?

SammyIAm, 2017-08-14:
No additional thought, but would prefer it be treated as CC-Attribution (will leave this issue open and look into adding language to the read me perhaps)

dmadison, 2017-08-14:
You can add the full text of the CC-Attribution license to the repository as a LICENSE file. The plaintext versions are on the Creative Commons website [here](https://creativecommons.org/2014/01/07/plaintext-versions-of-creative-commons-4-0-licenses/).

ocelotsloth, 2018-01-12:
@SammyIAm the CC license really isn't as well suited for software as some of the other options out there.

If you simply want the features that CC-BY provides, may I recommend using the MIT (https://choosealicense.com/licenses/mit/) license? It allows for distribution and modification, so long as attribution and copyright notice is provided. 

The important thing to take into account is that without *any* license provided, this repository falls under exclusive copyright, which is really quite restrictive. The importance of explicitly distributing a license along with the source code cannot be understated, especially if you intend for this to be free software.

To add the MIT (or other license) you just need to add a file named "LICENSE" to the root directory of the repository with the following contents:

```
MIT License

Copyright (c) [year] [fullname]

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.
```

Lothean, 2018-03-09:
@SammyIAm I do think that the project requires a licence.. Not a lot of people will contribute if they see that they have no rights with their own code.

dmadison, 2018-03-09:
It's actually worse than that @Lothean. Without an open source license, default copyright law applies. IANAL, but from my understanding this means that people contributing _do_ have rights to the code that they contribute. That's a problem, because it means that @SammyIAm, alone, cannot change the license without the permission from each contributor to license their code. This gets harder as the project moves along and more individuals add on to the codebase.

> When you make a creative work (which includes code), the work is under exclusive copyright by default. Unless you include a license that specifies otherwise, nobody else can use, copy, distribute, or modify your work without being at risk of take-downs, shake-downs, or litigation. Once the work has other contributors (each a copyright holder), “nobody” starts including you.

Source: https://choosealicense.com/no-permission/

The fact of the matter is that without a license, this is **not** free software.

SammyIAm, 2018-05-08:
Moppy 2.0 will be licensed under the Apache 2.0 license:
https://github.com/SammyIAm/Moppy2

This repo is unlikely to see additional commits and will not have a license added.

closed 2018-05-08 by SammyIAm