samphonic/MoppyClassic #170: error: cannot find symbol

Gaborboy95, 2017-07-08:
So.. i tried to create floppy music but this one problem keeps coming and i can't do anything. Tried to read some other topics, tried to fix with those, no succes. So i'm now making this issue report to help me.

Here's the error log:

ant -f C:\\Users\\peter\\Documents\\NetBeansProjects\\MoppyDesk -Dnb.internal.action.name=rebuild clean jar
init:
deps-clean:
Updating property file: C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\build\built-clean.properties
Deleting directory C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\build
clean:
init:
deps-jar:
Created dir: C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\build
Updating property file: C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\build\built-jar.properties
Created dir: C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\build\classes
Created dir: C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\build\empty
Created dir: C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\build\generated-sources\ap-source-output
Compiling 23 source files to C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\build\classes
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:3: error: cannot find symbol
import gnu.io.NRSerialPort;
  symbol:   class NRSerialPort
  location: package gnu.io
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:22: error: cannot find symbol
    NRSerialPort com;
  symbol:   class NRSerialPort
  location: class MoppyCOMBridge
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:194: error: cannot find symbol
            poolingControls1.connected();
  symbol:   variable poolingControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:195: error: cannot find symbol
            filterControls1.connected();
  symbol:   variable filterControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:217: error: cannot find symbol
            currentInputPanel.getTransmitter().setReceiver(filterControls1.getNoteFilter());
  symbol:   variable filterControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:220: error: cannot find symbol
            if (poolingControls1.isPoolingEnabled()){
  symbol:   variable poolingControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:221: error: cannot find symbol
                filterControls1.getNoteFilter().setReceiver(poolingControls1.getDrivePooler());
  symbol:   variable poolingControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:221: error: cannot find symbol
                filterControls1.getNoteFilter().setReceiver(poolingControls1.getDrivePooler());
  symbol:   variable filterControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:222: error: cannot find symbol
                poolingControls1.getDrivePooler().setReceiver(app.rm);
  symbol:   variable poolingControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:224: error: cannot find symbol
                filterControls1.getNoteFilter().setReceiver(app.rm);
  symbol:   variable filterControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:242: error: cannot find symbol
        poolingControls1.disconnected();
  symbol:   variable poolingControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:243: error: cannot find symbol
        filterControls1.disconnected();
  symbol:   variable filterControls1
  location: class MoppyControlWindow
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:26: error: cannot find symbol
        com = new NRSerialPort(portName, SERIAL_RATE);
  symbol:   class NRSerialPort
  location: class MoppyCOMBridge
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:108: error: cannot find symbol
        return NRSerialPort.getAvailableSerialPorts().toArray(new String[0]);
  symbol:   variable NRSerialPort
  location: class MoppyCOMBridge
Note: C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\PlaylistControls.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
14 errors
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\nbproject\build-impl.xml:923: The following error occurred while executing this line:
C:\Users\peter\Documents\NetBeansProjects\MoppyDesk\nbproject\build-impl.xml:263: Compile failed; see the compiler error output for details.
BUILD FAILED (total time: 0 seconds)

Thanks

Gaborboy95, 2017-07-09:
please help

SammyIAm, 2017-07-10:
It looks like your project isn't correctly/successfully loading the NRSerial libraries.  Make sure those have correctly been added to the classpath for the Moppy NetBeans project.  (I don't remember / know how that works off-hand, but it should be the same as for adding any other library which you can use Google to find instructions for)

Gaborboy95, 2017-07-11:
Thanks, added it now it works

closed 2017-07-11 by Gaborboy95