samphonic/MoppyClassic #171: Arduino Compiling error.  HELP!!!!!

esheridan1, 2017-07-22:
Hi.  Can anybody out there help me with this Arduino project that I am trying to compile. I'm getting these errors and I really don't understand why.  It appears to be repeating itself.  I'm very new to Arduino programming. Any help would be much appreciated.

Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Uno"

WARNING: Category '' in library Arduino (Firmata) is not valid. Setting to 'Uncategorized'
libraries\Timer-master\Event.cpp.o (symbol from plugin): In function `Event::Event()':

(.text+0x0): multiple definition of `Event::Event()'

sketch\Event.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Event.cpp.o (symbol from plugin): In function `Event::Event()':

(.text+0x0): multiple definition of `Event::Event()'

sketch\Event.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Event.cpp.o (symbol from plugin): In function `Event::Event()':

(.text+0x0): multiple definition of `Event::update(unsigned long)'

sketch\Event.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Event.cpp.o (symbol from plugin): In function `Event::Event()':

(.text+0x0): multiple definition of `Event::update()'

sketch\Event.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::Timer()'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::Timer()'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::stop(signed char)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::update(unsigned long)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::update()'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::findFreeEventIndex()'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::every(unsigned long, void (*)(), int)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::every(unsigned long, void (*)())'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::after(unsigned long, void (*)())'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::oscillate(unsigned char, unsigned long, unsigned char, int)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::oscillate(unsigned char, unsigned long, unsigned char)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::pulse(unsigned char, unsigned long, unsigned char)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer-master\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::pulseImmediate(unsigned char, unsigned long, unsigned char)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

MoppyDoppy, 2017-07-22:
You have two timer libs.  One in your sketch folder and in library. So please remove one and retry. All class names must be unique. Including sketch folder and libraries. 
You have a pair of .h and c / cpp files.

esheridan1, 2017-07-22:
Thank you MoppyDoppy.  I removed the files from the sketch folder just leaving the .h files in a directory called "Timer" and a sub directory "src".  When i try to compile I now get the following error
C:\Users\User\Documents\Arduino\sketch_Web-Server_Data_Logger\sketch_Web-Server_Data_Logger.ino:44:80: fatal error: Timer.h: No such file or directory

I am really confused.

MoppyDoppy, 2017-07-22:
What do you compile?
Moppy.ino is your file.