samphonic/MoppyClassic #172: Multiple Floppies - Signal Bug

Skystriker, 2017-08-06:
Hi,

I noticed something strange as I added more floppies to my collection.
I have 5 hooked up in my current setup. As long as only one or two play simultaniously everything works fine. As soon as tree or more play together, only floppies 1 and 2 work. The rest stutters and barely moves.

Hoverver I checked each floppy. Alone they do their job perfectly fine. First I assumed it has to do with the power. I have a lab bench supply that counteracts voltage drops very quickly. So power shouldnt be the issue.

I guess that there is a problem with the signals from the arduino. But again there is no leaking from one floppy to another.

So I am out of ideas. Does anybody have the same issues or has an idea what the problem might be?

Greetings from Germany

AceoJack, 2017-08-07:
Hello there Skystriker!
The problem could be cause by the step and directional pins being switched. I had this problem early on. Check if the wires are switched at the arduino and the floppy drive.

Another problem that could cause this is the midi file. Check that the midi file has notes on all channels and that the midi has a third channel. 

In the tab were you upload the midi and connect the drives, make sure that the drives are connected properly and nothing has any errors. If you have errors pop up, take a screen shot and put them in this thread.

I hope some of this helps because I know how much it can be frustrating at times :D
-AceoJack

Skystriker, 2017-08-08:
Hi AceoJack,

thanks for the help. As I said there is no connection error. I made a midi file where all 8 channels hold the same note for a minute. If I connect only drive 1&2 in the applet everything works fine. Same with just 3&4 or just 4&5. If I connect three or more they stutter. No matter witch of the three.

The applet runs smooth and I can play files with no errors (Just 2 floppies).
So sadly no problems with power, wiring or midi.

Thanks again for the suggestions.

-Sky

SammyIAm, 2017-08-09:
Hey Sky, greetings from The States!

Stuttering as you add more drives is generally caused because either the serial line or the Arduino is getting saturated with note messages.  The default baud rate is a **very** conservative 9600, but if you've got a good serial connection you should be fine setting it to 115200 in the Arduino sketch and the Java code (in MoppyCOMBridge.java).

Give that a try, but I think that'll solve your problem (unless the MIDI you're trying to play is super-nuts, in which case you may need to try to simplify your MIDI a bit)

Skystriker, 2017-08-11:
Hi Sammy,

Greetings to the master himself. The problem seems to be connected to the baud rate.
Setting it higher did not help however, but changed the type of stuttering to a more radical stuttering. I dont think it can be described very well. I will upload videos of the floppies on different baud rates tomorrow as it is already dark outside.
The midi really is just a test musical scale for all 5 channels. So nothing fancy at all.

Till tomorrow

Sky

SammyIAm, 2017-08-11:
Ah, interesting.  The videos will definitely be helpful then.  Maybe also
upload the MIDI file here while you're at it in case there's something
weird there.

Skystriker, 2017-08-12:
Here is the comparison of different baud rates and their effect on the floppies: https://drive.google.com/open?id=0Bzm-u6fOCMHnNF9fU1VfaVU0eDg

I also attached the midi file: https://drive.google.com/open?id=0Bzm-u6fOCMHncVFDWmtfNnJzWkk

While compiling the Java code I noticed a warning. For the sake of completion I added it below.

Aug 12, 2017 6:00:01 PM java.util.prefs.WindowsPreferences <init>
WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.

SammyIAm, 2017-08-14:
I love videos, they make this so much easier. 😄 

Alright, to get it out of the way first, the warning is just a warning, no issues (at worst, Moppy might forget your settings between runs, but even with the warning I think it'll remember them).

The read-heads on your floppy drives aren't moving down their full length and back as intended  (the one on the far right looks closest to what you'd expect).  Whatever else is going on, the Arduino code will always count the number of steps forward or backward the head has taken and reverse at the ends.  Since that's not what we're seeing it means that the issue is either in the wiring between the Arduino and the drives, or the floppy drives themselves (baud rate isn't an issue, though since you've already adjusted it higher you could leave it there in case you have some crazy-fast MIDIs later).

Based on what I can see from the video image, my best guess now is that your drives don't have enough power.  Are you trying to power them through the Arduino, or are they directly connected to the power supply (you shouldn't try to power them through the Arduino)?  Either way, the wires running to the power pins look like they might be a little small, though certainly probably too small to power all the drives together (which may be why a couple drives work, but multiple drives don't).

Try either some heavier gauge wires for the power, or running those smaller wires from each drive directly to your power supply.

Skystriker, 2017-08-15:
The power is coming from a lab bench power supply (the one to the right in the video).
There is not much of a voltage drop noticable. The drives are in parallel get supplied from a rail on the project board.
[image]

SammyIAm, 2017-08-15:
It might still be worth experimenting with hooking each drive to the power supply directly on its own line just to rule this out.  Over that sort of distance, the voltage drop might not be much but it's possible there's too much resistance to meet the needs of the motors when they're all trying to do stuff at once.

(Still just a guess until tested of course, but it seems consistent with your symptoms)

MoppyDoppy, 2017-08-16:
Hi

Please use a normal pc power supply with constant power. I have a similar one and had the same problem. The peaks of switching on is to much for it. As the code starts from drive one in tick function problems starts at drive 3-5. 
make sure to connect Arduino ground with external power ground. 

About baudrate.
Midi with 3 Bytes/ command can send maximum 256 commands on 31250 Baud. Moppy use 2 Byte and maximum 16 commands at same time. 
Also it use hardware serial with buffer. 
9600 is enough. The arduino is close to maximum of his process power with the tick function. Setting the baud rate higher only stressing it while interrupting the process. 
If trying to use software serial it is to much. 

Regards
Moppydoppy

Skystriker, 2017-08-16:
Thanks for the suggestions. Sadly nothing seemed to work.
I additionaly tested:

-Directly connecting the floppies to the power supply
-Connecting Arduino Ground and Power Ground
-Wiring the floppy signal ground to power ground
- Using a computer PSU instead of the lab power supply
- Using thicker cables (for resistance)

For now I am really out of ideas what to test for

dmadison, 2017-10-09:
Responding to this late, but do you get the same results if you run the MIDI file for 5 floppies but only physically connect one? It's possible that there's a problem with the control board and it's getting bogged-down by the additional processing.

Related to that, if you have the hardware you could try using two Arduinos connected to the floppy drives on the same power supply (3 to one, 2 to the other). That should tell you fairly definitively if it's a power issue.