samphonic/MoppyClassic #176: Floppy drive working and clicking.

Renarix, 2017-10-03:
Hello. I have 4 floppy drive setup with arduino uno. And problem is: working and clicking. I have proper power source and it is properly grounded. Constrain notes is on. Im 12 and im new to it.
Also i have recommendation to put new feature- track duplication so i can play 1 track music on all floppy drives same.
Problem link: https://www.youtube.com/watch?v=00smzKDhzm4

MoppyDoppy, 2017-10-03:
Hi
It sounds as you have to much notes and to high notes on one track. 
Please remember: Only one note on one drive at the same time. Try to lower (transpose) all notes per track. 
Mapping in the software is one drive per midi track. 
Hint. Floppy drives can not play the right side of a piano. Only left side. 

Do not give up :-)

Renarix, 2017-10-03:
Ok. I have question- Is that edited midi file? - https://www.youtube.com/watch?v=LdgzsF_O7oI&t=80s

Renarix, 2017-10-03:
Which software you recommend for midis?

MoppyDoppy, 2017-10-03:
I’m using Anvil Studio free version. Nice software. 

I did the same with that sound
https://youtu.be/MWlx6n1SqrA

Just multiply the tracks. Midi have maximal 16 Tracks. Create only tracks as many you have drives. 

Maybe this software can also help you
https://www.tobias-erichsen.de/software/loopmidi.html
Anvil to loopmidi to Moppy midi in

Renarix, 2017-10-03:
Thanks! Is this ok? - https://www.youtube.com/watch?v=lnzK9n3eP7U
I used same midi and when i disconnect floppy drive 4 it sounds great. After connect 4 it sounds like in that video. I found that midi in github somewhere... :D

SammyIAm, 2017-10-03:
Hmm, interesting.  Thanks for the videos-- it makes things way easier to try to diagnose!

Since 3 drives works fine, but 4 doesn't with the same MIDI file, it seems like your MIDI file is probably not the problem.  (Plus, with constrain notes on, the notes should all automatically be transposed lower, so you don't need to worry about things being too high)

It's a little hard to tell from the video, but are all the drives getting power from the same set of wires?  And are they daisy chained, or do they each have their own connection to the power supply?  It could be that the 4th drive is taking just enough power that the motors in the drives aren't fully stepping when they should be.

Renarix, 2017-10-03:
Yes all drives are getting power from same set of wires. For power supply i have 40w charger that can supply 2.4 amp with usb and problem might be that... because one drive needs 960ma. But... that charger also has quick charge port so i will give it try. However tomorrow i will try to connect pc psu. Thanks for help. I have to go sleep lol.

SammyIAm, 2017-10-03:
Worth a try, though the quick-charge ports often need some special circuit negotiation to get the full wattage from them, so it might not work if you hook it straight up.  PC PSU is definitely the way to go.

If it's still not working, you might try powering the drives in pairs instead of using just one set of wires for all of them in case the smaller wires are causing some voltage-drop under load.

Sleep is your friend though, let us know how it works out when you try it.

Renarix, 2017-10-04:
Good news. Totaly fixed with psu :)

closed 2017-10-04 by Renarix

SammyIAm, 2017-10-05:
Yay!  Glad that fixed it.

Renarix, 2017-10-06:
Yeah it sounds so much better. Looking forward to get more drives.