samphonic/MoppyClassic #177: CD rom drive.

Renarix, 2017-10-06:
Is there any way to make it produce sound? It is ide interface.

DefectX11, 2017-10-06:
IDE stands for Integrated Drive Electronics, so basically the controller chips are integrated into the drive. Controls like stepping the head would be controlled on board, which would be difficult to break out. Also, CD drives don't need to move their head very much, and rely on a much greater precision than a floppy drive, so they're nearly silent. Maybe an older CD drive would make some noise, but not nearly at levels suitable for this project.

SammyIAm, 2017-10-06:
As @DefectX11 said, CD drives lack the super-handy interface that floppy drives have for directly controlling their stepper motors.  You definitely *could* get it to produce sound, but you'd probably have to bypass the CD drive circuitry and control the stepper motor directly.

Also as was said, they're a bit quieter than floppy drives, though they would probably be able to hit higher notes, so that's something...

Scanners and inkjet printers fall into this same category, by the way, of containing stepper motors, but controlling them via their own circuitry.

closed 2017-10-06 by SammyIAm