samphonic/MoppyClassic #181: Any idea what would cause my floppy drives to just not work after a few months of storage?

MuffyTarkin, 2017-11-21:
Some background, my setup involves 14 floppy drives, 1 desktop paper scanner, and 2 hard drives. Awhile back, I killed my Arduino UNO's by accidentally feeding them 24V into the microntroller pins (don't do that), and replaced them with Arduino 101's. It took a bit of fannagling but the only adjustment I had to make was replace all instances of TimerOne.h with CurieTimerOne.h and they worked flawlessly (and could handle more too, 32mhz compared to 16mhz and this was a noticeable improvement).

So, to cut to the chase, I decided to dust off the ol' drives and get everything set up to make more videos. The wiring system I have is basically set up in such a way that I never have to disconnect the drives from my arduinos, I just need to move the stacks together and plug in the power. Super easy. Relatively painless, sometimes my home-made FDD connectors come slightly loose and I have to push them back in, but otherwise, it's been solid and has never failed me until now.

Now, the scanner I use has the motor connected directly to its own controller that then connects up to the first Arduino, and this has never been an issue. It has always worked. So, I get everything set up, and when I load up the sketch for each Arduino, it goes through but the only device motor that resets is the scanner. Odd...none of the drives moved at all. I've changed nothing, so I figured maybe there's some funny wiring thing going on. I check out the wiring, nothing seems out of the ordinary, and I couldn't recall if every drive reset or if it was just the first one after I did the transition to the 101's so I tried playing some songs. Nope, only the scanner plays notes. No movement on any floppy drives! So I decided to thoroughly test each Arduino, basically going one drive at a time.

Well...two hours later and I can't find a single thing wrong with it. My floppy drives just will NOT move, but the scanner does...and that makes no sense to me. When I plug each Arduino in while the drives are powered on, I can hear the motors move as I fumble with the plug, so that tells me there are SOME signs of life...but I'm at a complete loss here.

I still have an old UNO that I use as a test unit...and hooking up to that, the drives do function, albeit horribly but that's because that test unit is also damaged but still somewhat functional. So, is this an Arduino problem? Is there any way I can confirm that without having to buy new ones? I'd hate to drop $60 on two replacement 101's and still have the same trouble I'm having now.

MuffyTarkin, 2017-11-21:
UPDATE:
As it turns out I'm just a moron. I've been using a breadboard for a very long time now to just hook in my amplifier and have a convenient way of splitting the output of it, and I also used it to connect one drive out of each Arduino group to a common ground between both Arduinos, and actually I recently replaced this breadboard with a protoboard and screw terminals to help keep my tiny chipamp from falling apart every time I moved the setup around. It seems I neglected to include a ground connection, and mistook the one coming out of my Arduino for the connection I run my scanner off of.

closed 2017-11-21 by MuffyTarkin

SammyIAm, 2017-11-21:
Heh, glad you found the issue!

MuffyTarkin, 2017-11-22:
Thanks! I have also learned you don't need to ground every drive, in fact you don't even need to ground any drive to the Arduino. If you connect the Arduino(s) to the ground of your power supply it serves the same purpose.