samphonic/MoppyClassic #182: Noisy Scanner Stepper Motor

MuffyTarkin, 2017-12-02:
I've been using a scanner with my setup for quite some time now, but I noticed that compared to lots of other videos featuring a scanner or large stepper motor of some kind, mine sounds awful in comparison. I believe this is because I likely have the completely wrong type of driver for this motor and I have ordered up some different ones to experiment with.

Sometimes it sounds great and has the type of sound I fully expect, a smooth kind of high pitched noise. But most of the time, when it changes notes it seems to have this very unpleasant grinding and squealing that accompanies it. Also, every once in awhile when the motor is halted between notes, some odd noise can be heard coming right from the motor itself. [Here's an example](http://www.mediafire.com/file/837y67hmtq36yj8/scanner.mp3) containing all of the problems I'm referring to. At the very start you can hear the strange noise I mentioned earlier, and as it plays notes you will hear the grinding and squealing. However, some of those notes also sound just fine.

Any ideas what might be causing this? The particular driver I'm using with this is one that's no longer available, but [this one](https://www.ebay.com/itm/Single-TB6600-Stepper-Motor-Driver-Controller-Micro-Step-CNC-Axis-2-4-Phase-New/272895288622?hash=item3f89d3a52e:g:9IIAAOSwYxBZ6jW7) I found is extremely similar. Mine is configured to 0.2A with a microstep resolution of 1/16th. I've experimented with all the different possible combinations, Setting it to a lower microstep resolution seems to make the issue go away, but then it doesn't hit as high of notes. It runs on 12VDC. For what it's worth, I know it isn't just the motor dying or going bad, I actually have a few other ScanJets that I've also used and the motors were still plagued with the same issues.

I've tried looking far and wide for a datasheet on the specific motor in this scanner, but all I'm able to source is a massive PDF containing information on a bunch of motors, none of which are even close to the one I have when it comes to the product number (sth-39d1126-02). The particular scanner is a ScanJet 5C, and it sounds substantially better if you put it in the maintenance test. The test makes it play a little jingle, and the motor sounds hundreds of times better when going through the driver designed for the motor rather than whatever one I'm using. Unfortunately interfacing with the driver the scanner came with is a lot more difficult than just hooking some wires into a pre-made driver board, so wiring into the existing driver is something I haven't tackled yet. I've perused the datasheet and can make some sense of what I'd have to hook into, but I was having trouble finding the pins for step and direction. If I can make this work with a 5v or 12v external driver I'd much rather do that.

ahnafr, 2017-12-10:
I have had the same issue but with my floppy drive.

Lothean, 2018-03-09:
Hello!
Scanner music is what I prefer.
I know your issue, and there's nothing you can do about it, like, really.
This is the lottery, some scanners sounds very good, and some completely sucks.
I got my hands on a HP ScanJet 5P and an Epson Perfection 1200U recently and they are reputed to be in the best scanners to make music ([the 5P already does music on its own](https://www.youtube.com/watch?v=NJQ2LpUYqqE))
The Epson Perfection 1200U is used in the floppotron and it's a very good scanner.
You can also search for the next scanners which are the bests but also very expensive and very rare (atleast in France) : HP ScanJet 3P, HP ScanJet 3C, HP ScanJet 4C, HP ScanJet 4P
All other scanners that I've used are total crap.
I previously used a DRV8825 board to control my stepper motors but they don't work well with these scanners.
I received 2 L298N H Bridge modules for my scanners and I hope it's gonna come out greate !
(BTW, I need help to adapt the Moppy code to make it work with my H bridge, the arduino code really confuses me, and after 2 days of trying, it still does not work.. and why the hell do I understand the Java code but not the Arduino one ? :( )
Anyway, hope I was useful!