samphonic/MoppyClassic #183: Second Arduino Issues

SammyIAm, 2017-12-05:
@Renarix said:
> I have another issue.... I got secound arduino today. Just uploaded code to arduino. Went in moppy program and for channel 11 switched to Com4 and nothing works. Light on arduino blinks. Wiring is correct because when i change channel 1 from Com3 to com4 everything works. Strange... Com3 is my main arduino.

SammyIAm, 2017-12-05:
Channel 11 will still be channel 11 when it arrives at your second Arduino, which means your Arduino *would in theory* try to play those notes from the drive connected to pins 22 and 23.  However out of the box, the code only supports 8 drives.

You can work around this by modifying the Arduino code on the second Arduino to subtract e.g. 20 from the pin information received over serial.  This should shift channel 11 (pin 22) down to channel 1 (pin 2).

Version 2.0 that I'm working on will include the ability to do this sort of channel shifting in the control GUI so you won't have to go changing as much code, but it's still a ways off from being generally usable.

Renarix, 2017-12-06:
Well... i will wait for moppy 2 because im too weak at programming..

MuffyTarkin, 2017-12-07:
@Renarix  Here is a modified Arduino code. Use this first sketch for your first Arduino (drives 1-8).

`#include <TimerOne.h>

boolean firstRun = true; // Used for one-run-only stuffs;

//First pin being used for floppies, and the last pin.  Used for looping over all pins.
const byte FIRST_PIN = 2;
const byte PIN_MAX = 17;
#define RESOLUTION 40 //Microsecond resolution for notes

/*NOTE: Many of the arrays below contain unused indexes.  This is 
 to prevent the Arduino from having to convert a pin input to an alternate
 array index and save as many cycles as possible.  In other words information 
 for pin 2 will be stored in index 2, and information for pin 4 will be 
 stored in index 4.*/

/*An array of maximum track positions for each step-control pin.  Even pins
 are used for control, so only even numbers need a value here.  3.5" Floppies have
 80 tracks, 5.25" have 50.  These should be doubled, because each tick is now
 half a position (use 158 and 98).
 */
int MAX_POSITION[] = {
  0,0,158,0,158,0,158,0,158,0,158,0,158,0,158,0,158,0};

//Array to track the current position of each floppy head.  (Only even indexes (i.e. 2,4,6...) are used)
int currentPosition[] = {
  0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0};

/*Array to keep track of state of each pin.  Even indexes track the control-pins for toggle purposes.  Odd indexes
 track direction-pins.  LOW = forward, HIGH=reverse
 */
int currentState[] = {
  0,0,LOW,LOW,LOW,LOW,LOW,LOW,LOW,LOW,LOW,LOW,LOW,LOW,LOW,LOW,LOW,LOW
};

//Current period assigned to each pin.  0 = off.  Each period is of the length specified by the RESOLUTION
//variable above.  i.e. A period of 10 is (RESOLUTION x 10) microseconds long.
unsigned int currentPeriod[] = {
  0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
};

//Current tick
unsigned int currentTick[] = {
  0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0 
};

//Setup pins (Even-odd pairs for step control and direction
void setup(){
  pinMode(13, OUTPUT);// Pin 13 has an LED connected on most Arduino boards
  pinMode(2, OUTPUT); // Step control 1
  pinMode(3, OUTPUT); // Direction 1
  pinMode(4, OUTPUT); // Step control 2
  pinMode(5, OUTPUT); // Direction 2
  pinMode(6, OUTPUT); // Step control 3
  pinMode(7, OUTPUT); // Direction 3
  pinMode(8, OUTPUT); // Step control 4
  pinMode(9, OUTPUT); // Direction 4
  pinMode(10, OUTPUT); // Step control 5
  pinMode(11, OUTPUT); // Direction 5
  pinMode(12, OUTPUT); // Step control 6
  pinMode(13, OUTPUT); // Direction 6
  pinMode(14, OUTPUT); // Step control 7
  pinMode(15, OUTPUT); // Direction 7
  pinMode(16, OUTPUT); // Step control 8
  pinMode(17, OUTPUT); // Direction 8

  //With all pins setup, let's do a first run reset
  resetAll();
  delay(1000);
[... 416 lines cut ...]
  //}

  //Stop all notes (don't want to be playing during/after reset)
  for (byte p=FIRST_PIN;p<=PIN_MAX;p+=2){
    currentPeriod[p] = 0; // Stop playing notes
  }

  // New all-at-once reset
  for (byte s=0;s<80;s++){ // For max drive's position
    for (byte p=FIRST_PIN;p<=PIN_MAX;p+=2){
      digitalWrite(p+1,HIGH); // Go in reverse
      digitalWrite(p,HIGH);
      digitalWrite(p,LOW);
    }
    delay(5);
  }

  for (byte p=FIRST_PIN;p<=PIN_MAX;p+=2){
    currentPosition[p] = 0; // We're reset.
    digitalWrite(p+1,LOW);
    currentState[p+1] = 0; // Ready to go forward.
  }

}
`

Renarix, 2017-12-09:
@MuffyTarkin thank you!

Renarix, 2017-12-09:
oh wait. When i try to upload you code it shows this error: Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\Renars\Desktop\arduino\UNO Moppy 1\Uno_Moppy_1\Uno_Moppy_1.ino:8:0: warning: "RESOLUTION" redefined

 #define RESOLUTION 40 //Microsecond resolution for notes

 ^

In file included from C:\Users\Renars\Desktop\arduino\UNO Moppy 1\Uno_Moppy_1\Uno_Moppy_1.ino:1:0:

C:\Program Files (x86)\Arduino\libraries\TimerOne/TimerOne.h:42:0: note: this is the location of the previous definition

 #define RESOLUTION 65536    // Timer1 is 16 bit

 ^

C:\Users\Renars\Desktop\arduino\UNO Moppy 1\Uno_Moppy_1\Uno_Moppy_1.ino: In function 'void setup()':

Uno_Moppy_1:74: error: 'CurieTimerOne' was not declared in this scope

   CurieTimerOne.initialize(RESOLUTION); // Set up a timer at the defined resolution

   ^

exit status 1
'CurieTimerOne' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

MuffyTarkin, 2017-12-09:
@Renarix ah crap, I thought I fixed that. I guess it didn't save. If you do a find and replace in both sketches, find "CurieTimerOnex and replace all with "TimerOne". That code was also modified for using Arduino 101's but I thought I edited it back to work with a regular Uno. Guess it didn't save. Weird.

Renarix, 2017-12-11:
There is something wrong in your code. When i try to play music head doesnt move back and forward. It just go in one direction. Original code works flawlessly

Lothean, 2018-03-09:
@Renarix [Should be fixed](http://www.mediafire.com/file/z9ntg29jup70btx/Modified_Moppy_Code.zip)