samphonic/MoppyClassic #185: Floppy drive select not working

ThatRandomGuy908, 2017-12-14:
So I am trying to get this to work and when I plug my wire into the ribbon cable and tap the wires together to activate the light its not turning on. I have tried both drive a and drive b mode. Could it be because the ribbon is twisted?

Renarix, 2017-12-16:
[image]
try this

ThatRandomGuy908, 2017-12-17:
@Renarix I tried what you said and it didn't light up (Although once the power supply turned off the floppy drive did make a faint sound) Attached is a picture of the ribbon cable I am using

[image]

SammyIAm, 2017-12-17:
Which wires are you tapping together?  Try connecting each pair of pins (1&2, 5&6, 7&8, etc.) to see if any of the pairs turn the LED on.  It's possible you're looking at the cable upside down.

ThatRandomGuy908, 2017-12-18:
@SammyIAm The notch is supposed to be facing down correct?

SammyIAm, 2018-01-02:
It's different for different drives potentially... The notch *should* be on the odd-numbered pin-side (but even that could be off for some reason).  If you're holding it notch-down and looking at the female end (like in your picture) the number 1 pin (again, *should*) be on the bottom right.

I'd still suggest just going down the pairs and connecting each of them until you find what you're looking for.