samphonic/MoppyClassic #187: Serial connect issue (libNRJavaSerial.dll)

rorrexatnys, 2017-12-23:
Hey all, so I've got my 'ol floppy drives ready and compiled MoppyDesk using NetBeans "Clean and Build Main Project". I can run the MoppyDesk JAR, however when I click "connect" in the MoppyDesk GUI, the program closes and produces the following error:

```
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00000000006c862e, pid=6236, tid=5380
#
# JRE version: Java(TM) SE Runtime Environment (9.0+11) (build 9.0.1+11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (9.0.1+11, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [libNRJavaSerial.dll+0x862e]
```
More info:
I do have nrjavaserial-3.93 in my "lib" folder where the MoppyDesk JAR is.
I tried putting nrjavaserial in my JRE "lib" folder.
I also tried putting nrjavaserial in ' netbeans_jdkhome="C:\Program Files\Java\jdk-9.0.1" ' (from NetBeans conf. file).
Running Windows 10 64-bit.
I'm very novice with Java and NetBeans.

Any help is greatly appreciated!

rorrexatnys, 2017-12-24:
Nevermind, turns out a re-installation of JDK (version 8) and NetBeans fixed it.

SammyIAm, 2018-01-02:
Glad you got it fixed!

closed 2018-01-02 by SammyIAm