samphonic/MoppyClassic #188: Utilize headbanging

redcubie, 2018-01-04:
Some songs need headbanging to sound right.
You should add a "*Allow headbanging*" option to your `MoppyDesk` application.
I don't think it's possible to add this without modifying the `Arduino` code.
But the `Java` code is probably "not so easy" because you would have to make a trigger mechanism for when to *headbang*.

SammyIAm, 2018-01-04:
I think really up to the listener to decide if they want to headbang along to the floppy music or not. ;P

(What's "headbanging" in this context mean?)

redcubie, 2018-01-05:
I mean you should add an option to enable it easily, like a checkbox saying "*Allow headbanging*" in `MoppyDesk` to allow it.

(By headbanging I mean that the head "*bangs*" against the back of the drive.)
Also, good joke. :)

SammyIAm, 2018-01-06:
Ah, okay, thanks for the clarification.

What I was trying to avoid was the sort of clacking/grinding noises you can hear [in this video](https://www.youtube.com/watch?v=wjfkcdzPpLw) at around 27 seconds (it was worse in this particular video because I'd stood the drives up vertically, and they didn't like that).

Basically, sometimes the Arduino tells the drive to take a step, and for whatever reason it doesn't actually take that step (but the Arduino code still tracks that step as having happened).  The result is that when the head reaches its maximum position, the Arduino will tell the motor to keep on going (resulting in various hardware bits crashing and grinding).  To be clear, the 158 means the head will stop one step from the *center* of the drive-- it will always return to "0" at the back of the drive.  At the back of the drive, most drives have a little limit switch that will keep the head from moving too far in that direction.

Can you give me an example (video? audio?) of the sort of sound you're looking for?  I'm not sure I'm seeing which songs would benefit from randomly-sprinkled grinding noises :P

redcubie, 2018-01-07:
Not all drives have limit switches! Old computers utilized headbanging as a "*zeroing algorithm*" when booting. I think some retro computer games used headbanging as "copy protection".
Also I mean this kind of headbanging: [https://youtu.be/7EwU-juNl1M?t=16s](url) (This channel has nothing to do with me but he has such a magnificent PC hardware orchestra.)

SammyIAm, 2018-01-08:
Hmm, I'm actually not even sure there's headbanging going on in that video-- it sounds like it could just be the drives taking one step at a time (like the rhythm section in [this video](https://www.youtube.com/watch?v=w4gVlhjXWAo).

However, on the assumption that those drives *are* trying to step back past their limits and this is a feature that you want: The sound you're looking for is a sort of rhythm section thing rather than pitches, which would involve a bit more than just a checkbox.  Probably a good approach would be to optionally take note events from channel 10 (the default MIDI drum channel) and play them as a single step on the drives (always toward the back of the drive) rather than interpreting the notes as pitches.  You might take a look at some of @sparklini 's changes to drive relays, since the code would be very similar.