samphonic/MoppyClassic #191: MIDI Keyboard not working

asql2580, 2018-01-29:
I currently have 9 drives hooked up to an UNO that work great when playing a MIDI file. I can also play them directly from my computer (with a virtual MIDI keyboard and port). But when I try to use my keyboard (with USB-MIDI adapter), the drives won't play. However, the RX light on the uno flashes whenever I press a key; I just don't get anything from the drives. Any ideas as to what is going on? The keyboard and adapter work fine with Ableton Live, so I'm pretty sure the issue isn't there...

SammyIAm, 2018-01-30:
The RX light flashing is a good sign that things are at least mostly hooked up.  My best guess is that your keyboard is playing in too high an octave for the drives.  If your keyboard has an octave adjustment, try pulling it down about two octaves, otherwise you could also give the "Constrain Notes" checkbox a try to see if that pushes them into a usable range (though you'll be limited to only the drives' upper octave).

SammyIAm, 2018-06-04:
[Moppy2](https://github.com/SammyIAm/Moppy2) now has MIDI support, so it's worth trying again there.  I still suspect it's an octave issue though, so adjusting they keyboard to lower octaves should work.

closed 2018-06-04 by SammyIAm