samphonic/MoppyClassic #192: Scanner Motor / Arduino

Mason10198, 2018-02-03:


Lothean, 2018-03-09:
Hello, 
In the Arduino code, change this
`byte MAX_POSITION[] = {
  0,0,10000,0,158,0,158,0,158,0,158,0,158,0,158,0,158,0};

//Array to track the current position of each floppy head.  (Only even indexes (i.e. 2,4,6...) are used)
byte currentPosition[] = {
  0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0};`
to this : 
`int MAX_POSITION[] = {
  0,0,10000,0,158,0,158,0,158,0,158,0,158,0,158,0,158,0};

//Array to track the current position of each floppy head.  (Only even indexes (i.e. 2,4,6...) are used)
int currentPosition[] = {
  0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0};`

A byte number can't go higher than 255, so you were stuck no matter the number!
By the way, which scanner do you use ?

Mason10198, 2018-03-09:
WOW, of course. Thank you, I totally missed that.
I grabbed 3 different Visioneer models off of ebay, and used the internal board of one of them to make an interfacing board for the motors, since they all use the same type of motor.

Lothean, 2018-03-09:
Nice! Would love to know how they sound :)
And sorry for the late reply ;)