samphonic/MoppyClassic #194: What am I doing wrong?

CCF100, 2018-03-06:
Didn't say in video: ground pins are shorted together.
https://youtu.be/erjAiT7MRSE

Lothean, 2018-03-09:
Courtesy please! So hum... **Hello**
The only thing that you need to do, is to make a common ground (link the ground of the fdd to the Arduino)

Chong-McBong, 2018-03-12:
hopefully this pic helps [url]https://imgur.com/gallery/Vfihi[/url]

if you split open the connector on a floppy cable, you can jam your wires in the pins, and glue the connector back together, it works nicely on my setup, and saves soldering anything onto your drives, so they can be easily swapped

adaringdesign, 2019-11-17:
> hopefully this pic helps https://imgur.com/gallery/Vfihi
> 
> if you split open the connector on a floppy cable, you can jam your wires in the pins, and glue the connector back together, it works nicely on my setup, and saves soldering anything onto your drives, so they can be easily swapped

This was so helpful, thanks!