samphonic/MoppyClassic #195: L298N H Bridges, 2 good scanners but no skill

Lothean, 2018-03-09:
Hello!
Got my hand on some really gooood scanners in term of music (ScanJet 5P and Epson Perfection 1200U, which of the scanners used in the floppotron) and I decided to use L298N H Bridges modules to control them (I didn't liked the behaviour of the DRV8825/A4988 boards for Arduino)
Can someone help me figuring out what do I need to to with the Arduino code to make them work ?
So far, I successfully made them work with 2 arduinos, one with moppy and one that just have a code which gets informations from a Moppy pin on the other arduino and that makes a step on the motor everytime it goes from HIGH to LOW.
Any information is welcome!

MoppyDoppy, 2018-03-10:
Hi

I would like to help you and get also a scanner running. I will try to get the Bridge and a Scanner. (Epson?)
Can you provide information how to connect arduino to the bridge to the scanner and your special code of it?

Thanks

Lothean, 2018-03-11:
After all, I suggest you to get a DRV8825 or a A4988 board because it's much simpler. It turns out that the L298N is a pile of crap according to modern standards. (High voltage drop and oh god the heat... I tried to cook bacon on it and it worked..)
I personnally had one DRV8825 laying around and used it for my 5P scanner and it's way better. I'm gonna order a second one for the Epson one. They are a little bit tricky to use because you absolutely need to set the current and they are easily damageable. (Believe me, I bought 4 of them and the one I have is the only one left but at least I know what not to do). If you need any help on using these boards, tell me but I don't reccommend you these L298N, they suck.
For your scanner, Pawel Zadrozniak (the guy who made the floppotron) told me that "anything with large read head sounds OK", but he is wrong. In fact, only some stepper motors will work correctly. 
I'm providing you a confirmed list of what works great (they all have the same sound, if you need some samples, tell me) : HP ScanJet 3P, HP ScanJet 3C, HP ScanJet 4C, HP ScanJet 4P; HP ScanJet 5P, Epson Perfection 1200U
I've been looking for good scanners since maybe a year now, and after trying a lot of them, I can tell you that only the ones on this list have no issues. You just need to use them correctly with a DRV8825 board that you configured correctly.
Although these scanners are pretty expensive on eBay, and pretty rare (atleast in France), you can hunt on some "offers sites" (correct me if that's not the right way of saying this). I got a bargain : for 3€ I got a 5P and an Epson Perfection 1200U ! 
Good luck and I'm here for any help!

Lothean, 2018-05-21:
Solved with Moppy2

closed 2018-05-21 by Lothean