samphonic/MoppyClassic #196: Moppy And Bluetooth

CamoJackson, 2018-04-25:
I am trying to hook up an HT-06 bt to serial adapter and moppy isn't seeing the port. arduino can see it and write to it. any thoughts.

GeorgeTsakiridis, 2018-04-28:
Yesterday I received my HT-05 Bluetooth module and I wanted to test it with the Moppy. It worked ok. Moppy can't detect a bluetooth module because it is not designed to. It searches for COM ports. So what you need to do is modify the source code to search for bluetooth modules. This is actually pretty easy. You can download "BlueCove version 2.1.1-SNAPSHOT" library (assuming you have 64bit windows) and import it to the project. You need to write your own code for getting the device. Here is mine:

Bluetooth.java class: https://pastebin.com/QTcTMNL3
BluetoothScan.java class: https://pastebin.com/fPStecFd
Modified MoppyCOMBridge.java class: https://pastebin.com/mZdmeHKK

Note that this code was written in a few minutes and it causes some issues. For the program will no longer work with COM ports only with BT, and you must restart the program every time you with to disconnect. Also when the program launches you should just press Connect. Even if the selected COM port seems empty. Hope that helped!

SammyIAm, 2018-04-30:
Alternatively you should also be able to configure your Bluetooth device with a COM port of its own so it will show up with the other COM ports ([this page](http://www.collideabq.com/2016/04/28/bluetooth-serial-connection-with-windows-10/) has some instructions).

I've added directly bluetooth support as an [enhancement ticket](https://github.com/SammyIAm/Moppy2/issues/5) for Moppy2 though, perhaps someday it'll get added natively.