samphonic/MoppyClassic #201: Having issues with midi channels

Cont3mplation, 2018-07-03:
So I found an old floppy drive last night and immediately knew what to do with it, and having set up the moppy software and connected it to the arduino it works well and I can play midi files from it. However, when trying the example music tetris the drive only plays the second channel/bass notes that are under the melody (when channel 1 is selected on the right hand side) and seemingly no matter what other channels I select it can either play only the bass or its just silence. This may be me using the moppy software wrong as I'll admit I'm pretty new to it and my understanding is limited, but does anyone know what's causing this/how it can be fixed? Thanks in advance :D

Lothean, 2018-07-03:
Hey! 
You need to use the mapper, which is available in the new version of Moppy, called Moppy2
https://github.com/SammyIAm/Moppy2
With the new Moppy2, you can select which channel of the MIDI file plays on drive 1
Good luck!

Cont3mplation, 2018-07-06:
Hey thanks man, will look into it :D