samphonic/MoppyClassic #202: Can't select channels -- Interface issue?

OdinMinusNull, 2018-07-03:
Hi, so there was an issue like this posted two years ago that didn't get an answer. I'm trying my luck now, as I seem to have a similar problem:
I've read about Pooling and Round-Robin and wondered where I could select these options. Apparently, they're not showing up in my interface. Instead, I get two warnings when I start up Moppy:

```
Jul 03, 2018 1:47:54 PM org.jdesktop.application.LocalStorage getId
WARNUNG: unspecified resource Application.id using MoppyUI
Jul 03, 2018 1:47:54 PM org.jdesktop.application.LocalStorage getId
WARNUNG: unspecified resource Application.vendorId using UnknownApplicationVendor
```
The project itself works and the drives are playing their channels stubbornly when I feed them a MIDI. However, I'm generating a file in another project that only has one channel, but I'd love to use the Round-Robin option with it so it's not only one of six floppies doing something.

How can I get my interface to work properly? What's the issue? Thanks!

Lothean, 2018-07-03:
Hello! 
The original Moppy has been replaced by Moppy2, so it would make more sense to ask for a round robin on Moppy2 rather than trying to debug this one 
Here's the moppy2 repo where you can find the newest version and where you can open an issue suggesting round robin :
https://github.com/SammyIAm/Moppy2
Have a nice day!

OdinMinusNull, 2018-07-03:
Hey, thanks! 
I've tried to come by with the original Moppy, as I've found more tutorials for it, but oh well.
Anyways, I've tried my best to get 2.0 to work, but I seem to have made some mistakes there and now it doesn't work at all. I've opened an issue on the 2.0 rep, in case you want to help a total beginner out some more. :)

SammyIAm, 2018-07-03:
Left a response to your questions in 2.0-- hopefully we can get that working for you.  As @Lothean points out, there's no Round-Robin support in 2.0 yet, but I was thinking about it this morning after seeing your post and I've maybe got an idea for adding it.  Hopefully even soon :)