samphonic/MoppyClassic #203: Please Don't Delete Moppy 1.0 (b1e55ad)

dmadison, 2018-07-05:
I know Moppy 2.0 is planned to be an improvement, but Moppy 1.0 is still functional and good enough for many people's needs. Deleting everything from the repository is just going to make things frustrating for new users, likely linked here from one of many different tutorials across the web.

Even though the older commits are still accessible, this just makes things confusing for people trying to use the older version for whatever reason. A better option would be to leave the repo as-is and [archive it](https://help.github.com/articles/archiving-a-github-repository/).

At the very least, please add a link to the README for [accessing the older version](https://github.com/SammyIAm/Moppy/tree/4e771da2528f848401c2ec665ada77504d2d7b1a).

Lothean, 2018-07-05:
I disagree. New users will use Moppy2, which is a good thing since it's the only one that's supported.
Advanced users know what they're doing, so if they want to use a deprecated version of a software, they should have atleast the knowledge to revert a commit on a cloned repo.

SammyIAm, 2018-07-05:
Moppy 2.0 is significantly easier to setup and use, so I'd like to do whatever possible to direct new users to that one instead.  (It's also still compatible with the hardware setups from any previous tutorials, so no one should get totally left out just because they're following an old site)

I agree that the current setup is slightly more confusing than I'd like it to be (ideally there would be a way for me to automatically redirect people from this repository to the new one), but even after trying to make a clear note in the README, I've still been getting new users trying (and being unable) to get the older Moppy to work.  Removing everything but the README is the clearest way I can find to make sure people are using 2.0 instead.

I will look into marking this repo as archived, and might see if Github will be nice and do a one-off redirect for me :)

closed 2018-07-05 by SammyIAm