samphonic/MoppyClassic #207: Floppy do not working

LucasLePompier, 2024-01-10:
Hello,
I have floppy disk arduino uno and ATX power supply.
The program on the arduino is working (tested with stepper motor and A4988 driver with Moppy Classic !)
and my floppy disk is powered and if I start music by the sample song of the moppy 2.0 (only for the sample song I use moppy 2.0) the floppy drive not working...

I use a TEAC FD 235 HG and the step pin is connected to the pin two of the arduino and dir pin to the pin 3... and if I reverse the pin 2 and 3 and I press reset button on the arduino the stepper motor do 1 step...

Sorry for my English I'm French and Teen !

Sammy1Am, 2024-01-10:
Hi there,

MoppyClassic isn't really being maintained or supported anymore, so my first suggestion would be to go install the Arduino code from [Moppy2](https://github.com/Sammy1Am/Moppy2).  One of the advantages of the new code is it will play a little startup tone on the drive to confirm the Arduino and floppy are correctly connected, which is a very helpful troubleshooting step.

Is there a specific reason you're trying to use MoppyClassic instead?

LucasLePompier, 2024-01-11:
Hi
I dont use moppy 2 because the .bat files start and quit immediatly !

LucasLePompier, 2024-01-11:
And for the pin dir and step the arduino difusse gnd or 5V ? (My floppy drive only accept GND to pin dir and step I think )

LucasLePompier, 2024-01-12:
But now is the arduino sketch...

Sammy1Am, 2024-01-12:
Yeah, have you uploaded the Arduino sketch from Moppy2?  That has the
startup tone, but also is required to work with the Moppy2 GUI.

LucasLePompier, 2024-01-12:
But I have uploaded the sketc to the arduino but arduino IDE makes an error...

Sammy1Am, 2024-01-12:
What's the error?

LucasLePompier, 2024-01-13:
[image]
This is the error !

LucasLePompier, 2024-01-13:
The problem is solved ! after an new version of the files !

LucasLePompier, 2024-01-13:
but now how can i makes the led of the floppy drive lights up when it has an action otherwise is off ?

LucasLePompier, 2024-01-15:
Hello ? 
How I can makes the led blinking when the stepper move and led stop on the stepper stop ?

Sammy1Am, 2024-01-15:
Greetings!

There's a couple options:

1. The LED on the drive turns on when the `DRIVE SELECT` pin is connected to ground.  The normal instructions just have you do this all the time, but if you'd like to control that with the Arduino, you'll need to connect that pin from the floppy to the Arudino instead.  You'd need to use a pin outside the range of the pins selected for use as instruments in Moppy (D18 is the first available on the Arduino Uno by default, but if you're using less than 8 drives you can adjust [this line](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L41) and it should let you use lower numbers for the LED.
Then you just need to modify the code to toggle on and off the select PIN whenever notes are playing.  It'll need to be on all the time you want notes because otherwise the drive is unselected and won't respond to input.  Probably easiest to just shove it in [FloppyDrives.cpp](https://github.com/Sammy1Am/Moppy2/blob/master/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/FloppyDrives.cpp).

2. Because the DRIVE SELECT pin determines whether or not the drive will respond to input, if you're looking for something more purely display-oriented, you could just connect separate LEDs for each drive (or disconnect the LEDs on the drive and wire them up separately.  Then you could connect the LED to the `STEP` pin for the drive and it would automatically turn on whenever a note is playing, so you wouldn't need to make any software changes.

---

That should give you a good software (1) or hardware (2) option.  There's a more complicated 3rd option where you add software-controlled, separate LEDs (like in some of my [newer videos](https://youtu.be/onXwRfyqNP8)).  There's some code for this in the [experimental branch](https://github.com/Sammy1Am/Moppy2/tree/experimental/Microcontroller/Moppy2-Arduino/src/MoppyDecorations), but it's been long enough I don't remember exactly how it works.

LucasLePompier, 2024-01-15:
I use this option : "Because the DRIVE SELECT pin determines whether or not the drive will respond to input, if you're looking for something more purely display-oriented, you could just connect separate LEDs for each drive (or disconnect the LEDs on the drive and wire them up separately. Then you could connect the LED to the STEP pin for the drive and it would automatically turn on whenever a note is playing, so you wouldn't need to make any software changes."

The LED shines even when it is not playing notes and once the music starts it shines when it should! (this english is google translate)

LucasLePompier, 2024-01-15:
And if he move big steps the led stop shines !

LucasLePompier, 2024-01-18:
Hello,
How can I use a stepper motor for drums? ?
And can i use custom board for Moppy ?

Sammy1Am, 2024-01-19:
> How can I use a stepper motor for drums? ?

Er, I'm not sure.  Most drum setups I've seen have used something more like a relay/solenoid sort of thing for percussion.  If you wanted to do that, you could use a shift register and switch to the ShiftRegister instrument by commenting/uncommenting the [appropriate lines here](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L20).  Each position on the shift register will be mapped to a specific MIDI note.

I guess if pressed you could use a stepper motor by attaching several drumsticks to it radially and turning it one-stick-worth each time the note was activated.  You'd need a new instrument class to handle that use case though.

> And can i use custom board (with atmega328p) for Moppy ?
Yeah, sure.  Anything with support for the Arduino framework should be fine (I've used ESP8266s successfully).

LucasLePompier, 2024-01-20:
Hello,
Can i use cd reader for this ?

Sammy1Am, 2024-01-21:
Yes.  Basically anything electronically controlled can be used with this.  **However**, the CD drive interface is not as straightforward as the floppy drive, so you'd most likely need to use a controller to hardwire directly to the stepper motor(s) in the CD drive.